tommag/TMC5160_Arduino #1: type_traits not found during compile

phodgers, 2021-01-06:
Hello,

I tried to compile the SPI example on Arduino 1.8.13 but found an issue with type_traits not being found. I installed ArxTypeTraits from Arduino Library and referenced that in bitfield.h and it then compiled okay.

tommag, 2021-01-06:
Thanks for reporting ! I should probably include ArxTypeTraits directly.

matt-lamport, 2021-05-10:
I encountered the same issue when trying to compile the SPI example on Arduino 1.8.8 on Windows 10. All I had to do was change Line 11 in Bitfield.h from "#include <type_traits>" to "#include "type_traits.h" in order to get it to compile properly.

tommag, 2021-05-24:
ArxTypeTraits is now a library dependency for Arduino and PlatformIO.

closed 2021-05-24 by tommag