tommag/TMC5160_Arduino #2: Getting SPI example to work on TMC5160-BOB

phodgers, 2021-01-06:
Hello,

I am trying to get up and running with TMC5160-BOB using the SPI example. The break out board https://www.trinamic.com/support/eval-kits/details/tmc5160-bob/ comes pre-configured for SPI comms (so 'Tie SPI_MODE to VCC_IO, SD_MODE to GND' should be taken care of). Also I have tied CLK16 input to GND.

I ran the SPI example and it seems to configure the TMC5160 correctly, because it energises the motor and holds it in place, but I'm not getting any movement. I am not using any encoder, but I don't think that matters?

When I check the serial output it just streams the same output of
"current position : 0.00	current speed : nan"

Any ideas of where I might be going wrong?

Thanks.

tommag, 2021-01-06:
Hello, 

You don't need an encoder.

Which microcontroller board do you use ? I have written this example yesterday and tested it with a Teensy 3 and a TMC5160-BOB.

tommag, 2021-01-06:
I assume that you used a AVR based board, I rarely use these and hadn't tested the lib with one of those boards. It turns out that I had already fixed issues with AVR boards in https://github.com/tommag/Estee_TMC5130/

Please tell me if the latest commit fixes the issue.

phodgers, 2021-01-06:
Yes it works now! I am using an Adafruit ItsyBitsy (32u4).

Thanks.

tommag, 2021-01-07:
FIxed in #d220fb5

closed 2021-01-07 by tommag