tommag/TMC5160_Arduino #3: Motor Not Moving

tseanard, 2021-02-06:
I am running the main TMC5106_SPI.ino script on a Teensy 4.1 over SPI.  It is talking over SPI, and the position/speed is being reported as changing in Serial Monitor (Arduino 1.8.13, Windows 10, Teensyduino Teensy Loader 1.53)  I have the 4 SPI pins, enable pin, VCC_IO, gnd, VM, gnd all connected, as well as the 4 motor pins.  The motor is not moving.

tommag, 2021-02-06:
Hello,
Which driver board do you use ? Can you send a schematic of the connexions ? 

Is the motor energized on startup or does it move freely ? 

The fact that the position & speed vary in the serial monitor seems to indicate that the SPI bus is working correctly, and the driver clock as well. Maybe it is a problem with the TMC5160 DRV_ENN line ? You can tie it to GND if not needed.

tseanard, 2021-02-06:
The board is BigTreeTech TMC5160 v1.2 (https://www.biqu.equipment/products/bigtreetech-tmc5160-v1-0-driver-spi-mode-silent-high-precision-stepstick-stepper-motor-driver-with-heatsink-for-skr-v1-3-gen-v1-4-reprap?variant=20225467711586)

I have tried pulling ENN to ground via jumper, and that doesn't seem to help.  The motor is getting current (it's held in single position more strongly than when disconnected), it's just not moving.

If there isn't something that you have seen already, I'll dig in to the datasheet for the TMC5160, but I wanted to check first if there was something obvious I missed.

tommag, 2021-02-06:
It looks that on this driver board the "motion controller" mode is disabled (i.e. it expects Step/Dir signals). 
As seen here there seems to be a workaround : https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/issues/8 by cutting a trace on the PCB. 
Let me know if this solved the issue for you !

tseanard, 2021-02-07:
I do believe that cutting that trace will do the trick, but it's too small for the tools I have.  I'm going to use a different board instead.

closed 2021-02-07 by tseanard

alexander-sorokin, 2023-06-19:
I have the same issue.
My board is BigTreeTech TMC5160 v1.3.
When I am disconnecting ENN pin and holding wire by hands, motor starts ticking in random directions. When ENN pin grounded driver supplying holding current but does not move motor. SPI example is running and reports position changes wile I am connecting/disconnecting ENN pin.

tommag, 2023-06-22:
Well.. Have you read the discussion and tried the proposed fix ? It looks like it's much easier on the v1.3 board : https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/issues/8#issuecomment-1370375614 (comment from the issue linked above).

alexander-sorokin, 2023-06-23:
Yes I tried. It does not help. Driver behaves the same way with resistor soldered on top pair, low pair of contacts or desoldered at all. I tried few different drivers and all of them are acting the same way. I also used BTT 1.2 version with cut trace from thread you mentioned. The same result.
Was anyone successful at using BTT TMC5160 in SPI or UART modes?