tommag/TMC5160_Arduino #7: Use library in own class: SPI on ESP32 fails (MOSI floating)

logowe, 2021-11-02:
Hello Tom
1) I have a 'working application' which uses TMC5160_Arduino successfully with 3 motors.
The hardware used is DFROBOT FreeBeetle Board32 (V4.0) along with 3 Trinamic TMC5160 BOB. 
Software is Arduino IDE version 1.8.15 with Espressif ESP32 Board package 2.0.0.
In my application I use copies of the same motor code with the individual settings per motor. That is not very sophisticated but was simple enough to get it all working.

2) Now I took a further step with an own 'TMC5160_Motor' class which should let me create as many motor objects as I need - and get rid of copies I have in use in the 'working application'. 
Therefore I needed to have an object of your TMC5160_Arduino class in my own 'TMC5160_Motor' class, of course.
2.1) This is where I run into a nasty problem:
Looking at the signals generated with the oscilloscope it shows that the SPI port initialisation is not done completely, that is, 

- the CS_SPI pin works 

- but the SPI is left floating. 

I have tried many options  to declare a motor object and also associate these to the SPI port using TMC5160_SPI function.
After hours of investigating, I decided to raise this issue.

However, even though I have written some classes using other classes successfully I would't call myself a specialist in this matter, so I may miss some important point here.

I attach the essential code for investigation, if you don¨t mind to spend some of your valuable time, thank you in advance.

[Motor_TMC5160_V0.106.ino.txt](https://github.com/tommag/TMC5160_Arduino/files/7460003/Motor_TMC5160_V0.106.ino.txt)
[TMC5160_Motor.cpp.txt](https://github.com/tommag/TMC5160_Arduino/files/7460004/TMC5160_Motor.cpp.txt)
[TMC5160_Motor.h.txt](https://github.com/tommag/TMC5160_Arduino/files/7460005/TMC5160_Motor.h.txt)

tommag, 2021-11-02:
Hello,

 I think you just need to add a call to SPI.begin() in your setup :)

SPI init is not done by the library.

Let me know if this works, and happy to know that my lib is useful to someone!

Cheers,
Tom

logowe, 2021-11-03:
Hello Tom
YES this was the problem, it works now!
I am a bit ashamed - should have found this myself!!

I had a look at all libraries available in GitHub a month ago, aftzer some tests I choose yours because it was the best for my purposes, as of :
1. being tailored just for TMC5160, 
2. use of SPI interface, 
3. the motor tuning functions.
So, the only missing function is stall detection! (Hmmm, I see in your code a TODO for this.)

Thank you for the valuable work you have done!

closed 2021-11-03 by logowe