tommag/TMC5160_Arduino #13: uart mode setTargetPosition not working

imBunyip, 2022-05-04:
hi,
im using your library for spi fine, but i have now connected it up in uart mode but the position when running your tmc5160_uart example only moves back and forth to 0-56 not 0 to 200, i have tested the config wizard and the board is found fine but it doesnt move as much as its suposed to.

i have tested and it works perfect with Estee_TMC5130_UART the example back and forth moves perfectly between 0 and 200.

any ideas what the problem might be?

for testing im using an arduino mega on serial1.
[image]

from my thoughts if the estee version works this should to,

tommag, 2022-05-05:
This is indeed very weird, it goes to -56 instead of +200. It looks like an overflow or something. 

I don't use AVR 8-bits processors such as Arduino Uno or Mega, and I know there has been some bugs with this platform before because the primitive types are different from the ARM or ESP32 processors I am used to. 

Do you have a board with another architecture (M0 / M4 / ESP8266 / ESP32 ) that you could test with ?

imBunyip, 2022-05-05:
got it working with a teensy straight away, thanks "case closed"

tommag, 2022-05-06:
Thanks for reporting, I'll leave this issue open until I get the opportunity to test with an AVR board, there is no real reason not to support them!

tommag, 2022-05-14:
Could you please check if the latest pull request fixed the problem on AVR platform for you ? I am traveling and have no access to any hardware. Thanks !

imBunyip, 2022-09-13:
hey ,me again, i see on the trinamic site you can also do daisy chaining in spi mode do you have an example and is your library compatible already?bassically using only one cs  pin would be hugely advantageous.

closed 2022-11-03 by imBunyip