tommag/TMC5160_Arduino #15: Delay between SPI transfers is too short

DeanIsMe, 2022-09-22:
In `TMC5160_SPI::readRegister`, a delay is added between the 2 transfers with nop statements.
```
#define nop() asm volatile("nop")
nop();
nop();
nop();
```
On Teensy4 @ 600 MHz 3 system clocks is only 6 ns.
The TMC5160 datasheet has "CSN high time" minimum of 2*tCLK+10 ns, which is about 176 ns for the 12 MHz internal clock.
Hence the 'nop' delay is not compliant with the CSN high time.

**Impacts of this:**
In the _second_ transfer, if you read the SPI_STATUS register like this:
`spiStatus = _spi->transfer(address);` 
You'll see that the `spiStatus` is erratic and inconsistent. Increasing the delay between the transfers fixes this.

tommag, 2022-12-22:
Indeed. 
I see that Teensy defines a _delayNanoseconds()_ function that would be useful : https://www.pjrc.com/teensy/td_timing_delay.html#delayMicroseconds but I don't know of a nice cross-platform way of handling such a short delay, do you have an idea ?

DeanIsMe, 2022-12-22:
Check out #21. It should solve the problem in a cross-platform manner.

tommag, 2022-12-27:
Thank you ! Before merging this one I want to take some time to test it out on different platforms. 
Also, it looks like it could be useful to have this code in a small, separate lib, it may be useful for other use cases. Do you mind if I package it in another lib ?

DeanIsMe, 2022-12-30:
That makes sense.
No problem - you may package it in another lib.

tommag, 2023-04-07:
I pushed some commits that should solve your issue but I don't have a Teensy 4 to test. I've tested it with a Teensy 3.5 and it seems to work but there was already a 1us delay between SPI transactions without additional delay. 

Cheers and thanks for reporting !

closed 2023-04-07 by tommag