tommag/TMC5160_Arduino #19: setTargetPosition

Cachago13, 2022-12-22:
There is a fault in this function (maybe it's intended but I down't think so).
please change the int datatype at position to a float.

void TMC5160::setTargetPosition(float position)
{
	writeRegister(TMC5160_Reg::XTARGET, (int)(position * (float)_uStepCount));
}

change to 

void TMC5160::setTargetPosition(float position)
{
	writeRegister(TMC5160_Reg::XTARGET, (float)(position * (float)_uStepCount));
}

So it's possible to use the whole 2^31 range instead of the +- 128 steps.
Described in the datasheet at s.41

[image]

tommag, 2022-12-22:
It works as intended in multiple projects by multiple users afaik. 

I don't see why int would limit to +/- 128 steps ? Have you noticed this in your usage ? On which platform ? 

If your platform converts _int_ to _int8_t_ (seems weird to me, I think even on AVR it's _int16_t_), you can try to cast to _int32_t_ instead. If it solves your problem then feel free to submit a PR, this code has been mostly tested on 32-bits microcontroller platforms (ARM or ESP32) and I know there are some issues on AVR for example. 

In any case you definitely don't want to send the binary representation of a float to a device that expects an integer.

Cachago13, 2022-12-22:
I'm not sure why there is this behavior. Im using it on a Arduino Mega an programming it with visual studio code.

closed 2023-04-06 by tommag