tommag/TMC5160_Arduino #22: Max Speed

Cachago13, 2023-02-02:
Hi,
it's not directly a driver problem i think.
I have a NEMA14 stepper and i can't go higher than 850 with the speed (acceleration is 1000).
Do you have some Know How about this?
Is it possible to go much higher? Is it a stepper problem or a software adjustment?

tommag, 2023-02-03:
Hello, 

Can you be more precise with your issue ? What do you mean by "can't go higher than 850" ? Do you ask the driver to go faster and it doesn't or does the motor stall physically ? 

If the motor stalls (and you drive it with its specified current of course), then it's underpowered for your application and the only thing you can do is switch to a more powerful one.

Cachago13, 2023-02-03:
Hello,

first, sorry for my bad explanation :)
I've a 17hs2408 Stepper motor.
[image]

At the moment i'm not able to go higher than 850 steps per second --> slightly more than 4 rounds per secon.
Is this just because of the physical properties from my motor or should it be possible to go higher?
Over 850 steps per second the motor just vibrates.

Thanks for your help!

tommag, 2023-02-03:
In my experience most motors, when they are correctly sized for the load, can run at up to 1000 RPM (ie 3333 steps/s). 
So in your case it's probably that the motor is too weak for the load you're trying to move. 

Or if there is no load connected to the motor, maybe it needs some inertia on the axle, you can try adding a balanced weight to the motor shaft.

tommag, 2023-02-03:
I'm closing this issue because it's unrelated to the TMC5160 library.

closed 2023-02-03 by tommag