tommag/TMC5160_Arduino #23: BitField library produces undefined behaviour

DeanIsMe, 2023-02-14:
The author of the BitField library [has stated](https://blog.codef00.com/2014/12/06/portable-bitfields-using-c11) that his library produces undefined behaviour.

> NOTE: There has been some discussion in the comments about the degree of portability of this technique. There seems to be a growing consensus that this is in fact undefined behavior. For system-level code, using techniques that are technically UB is pretty much standard practice, but I wanted to fully disclose that unfortunately, this technique is likely UB and it is possible that some compilers/architectures could emit code that has unexpected results.

I've had a lot of difficulty getting my system to work well, and I just realized that the core of my issues were this BitField library resulting in invalid writes to the TMC5160 registers! 
Like all bugs related to compiler optimization, it's difficult to reproduce.
However, I found that by removing compiler optimization in BitField.h, my motors started to behave normally!

```
#pragma GCC push_options
#pragma GCC optimize ("O0")
#include <BitField.h>
#pragma GCC pop_options
```

This is an incredibly pernicious issue because it can produce a wild range of outcomes and is very difficult to detect.

DeanIsMe, 2023-02-14:
I'm using VSCode PlatformIO, Teensy 4.0 & Teensy 4.1

tommag, 2023-02-14:
Wow. Thanks for chasing this one down, must have been really frustrating to debug. 
I plan to spend some time on this lib soon, I'll add what you suggest.

DeanIsMe, 2023-02-14:
Sounds good! Note that the solution is GCC-specific, so it won't work with other compilers. Although I've never heard of an arduino project using a non-GCC compiler, so maybe that's fine.

tommag, 2023-04-06:
Thanks again for your report and proposed solution.

closed 2023-04-06 by tommag