tommag/TMC5160_Arduino #24: daisy chaining drivers

imBunyip, 2023-02-25:
i converted uart_chained_drivers.ino to this tmc5160 library.

so far i can get it to find each device and switch the nao pin to find the next driver along the chain.
so far i only have 2 to test with but i am able to find all 2 addresses 254,253.
but i cannot for the life of me move each motors seperatly.

i am unable to run for example.

`    motors[0].setTargetPosition(dir ? 2000  : 0); 
      motors[1].setTargetPosition(dir ? -2000 : 0);  
`
 when running the code below both motors move the exact same, almost acting as one.
`

#include <Arduino.h>
#include <TMC5160.h>

const uint8_t UART_TX_EN = 8;  // Differential transceiver TX enable pin
const int MAX_MOTOR_COUNT = 2;

TMC5160_UART_Transceiver motors[MAX_MOTOR_COUNT];

int motorCount = 0;

void setup() {
  // init serial coms
  Serial.begin(115200);
  while (!Serial)
    ;

  // Init TMC serial bus @ 500kbps
  Serial1.begin(500000);
  Serial1.setTimeout(2);  // TMC5160 should answer back immediately when reading a register. was 1 before

  // status LED
  pinMode(LED_BUILTIN, OUTPUT);

  // TMC5160 research on bus
  Serial.println("Starting to look for TMC5160 \n");

  bool chainEnd = false;
  while (!chainEnd && motorCount < MAX_MOTOR_COUNT) {

    motors[motorCount] = TMC5160_UART_Transceiver(UART_TX_EN, Serial1, 0); //starts at 0 no nai pulled up.
    motors[motorCount].setCommunicationMode(TMC5160_UART::RELIABLE_MODE);

    TMC5160_UART::ReadStatus readStatus;
        TMC5160_Reg::IOIN_Register ioin = {0};

    // uint32_t gconf = motors[motorCount].readRegister(TMC5160_Reg::GCONF, &readStatus);

    switch (readStatus) {
      case TMC5160_UART::SUCCESS:
        {
          Serial.println("Found TMC5130 @ address ");

          //Addressing
          uint8_t address = 254 - motorCount;
          motors[motorCount].setSlaveAddress(address, true);
          Serial.print("New address : ");
          Serial.println(address);
          Serial.println();

          // This sets the motor & driver parameters /!\ run the configWizard for your driver and motor for fine tuning !
          TMC5160::PowerStageParameters powerStageParams;  // defaults.
          TMC5160::MotorParameters motorParams;
          motorParams.globalScaler = 40;  // Adapt to your driver and motor (check TMC5160 datasheet - "Selecting sense resistors")
          motorParams.irun = 14;
          motorParams.ihold = 12;

          motors[motorCount].begin(powerStageParams, motorParams, TMC5160::NORMAL_MOTOR_DIRECTION);

          //Motor init
          motors[motorCount].setRampMode(TMC5160::POSITIONING_MODE);
          motors[motorCount].setMaxSpeed(900);
          motors[motorCount].setAcceleration(1000);

          motors[motorCount].writeRegister(TMC5160_Reg::IO_INPUT_OUTPUT, 0);  //set NAO low
          motors[motorCount].resetCommunication();
          motorCount++;
        }
        break;

      case TMC5160_UART::NO_REPLY:
        Serial.println("No more TMC5160 found.");
        chainEnd = true;
        break;

      case TMC5160_UART::BAD_CRC:
        Serial.println("A TMC5160 replied with a bad CRC. Trying again.");  //TODO keep a count of failed attempts.
        motors[motorCount].resetCommunication();
        break;
    }
  }

  Serial.println("Starting up");
}

// periodic delay
uint32_t t_echo = millis();
uint32_t t_dirchange = millis();
bool dir = false;

void loop() {
  uint32_t now = millis();

  // every n seconds or so...
  if (now - t_dirchange > 4000) {
    t_dirchange = now;

    // reverse direction
    dir = !dir;
    for (int i = 0; i < motorCount; i++) {
      motors[i].setTargetPosition(dir ? 200 * (i + 1) : 0);  // 1 full rotation = 200s/rev
    }

    digitalWrite(LED_BUILTIN, dir);
  }

}`

im using teensy 3.6 in arduino 2 and a max458 module.
ive attached my schematic of the build, there are two boards that stack top and bottom.

[image]
[image]

tommag, 2023-02-28:
If you don't have a pullup on NAI you should use `setSlaveAddress(..., false)` ; check comment here https://github.com/tommag/TMC5160_Arduino/blob/d377b7a28272c09d8b8054dd5ee8b8deb737576e/src/TMC5160.h#L266 

Otherwise are you sure that your motors are wired in exactly the same way ? Could they be wired so that they turn in opposite directions ?

imBunyip, 2023-03-07:
i have tried debbuging more over the past days still no avail.
i have noted that on the tmc5130 SWSEL exsisted and from the comment here the same should apply to the 5160 yet i cant find SWSEL inside the 5160 datasheet and never added it to my schematic.
https://github.com/tommag/TMC5160_Arduino/blob/d377b7a28272c09d8b8054dd5ee8b8deb737576e/src/TMC5160.h#L316
is this just copied from tmc5130 and not adapted as i cannot find an equivalant need.

tommag, 2023-03-08:
You're right, there is no SWSEL on TMC5160, it's a combination of SPI_MODE = L & SD_MODE = L that enables UART mode. 

From the log you've posted earlier there's definitely a communication issue, you can see a lot of 'no reply' while trying to communicate with the first motor. 

I don't really see why you commented out` uint32_t gconf = motors[motorCount].readRegister(TMC5160_Reg::GCONF, &readStatus);`. 
Indeed if you don't initialize `readStatus` by trying to read a register it will always evaluate at SUCCESS...

Sorry but it requires a lot of time to debug someone else's work... even more so when it's derived from mine (how can i know exactly what you've changed and what you've kept ???)

I would suggest hooking up a logic analyser to the UART bus between the controller and first motor, and between the first and second motors to check what really happens on the RS485 bus. This will help you to see if there is a timing issue somewhere. 

And I think that you need to address first the issue of why the first motor doesn't reply. Maybe a NAI high/low issue.

tommag, 2023-03-08:
Also, reading again the TMC5160 datasheet, if you only need 2 motors : 

> 5.4 Addressing Multiple Slaves 
  ADDRESSING ONE OR TWO SLAVES
If only one or two TMC5160 are addressed by a master using a single UART interface, a hardware address selection can be done by setting the NAI pins of both devices to different levels.

Just set NAI (MOSI) on one of your devices to LOW, NAI on the other to HIGH, tie SWP and SWN of both drivers together with the RS485 output and you're done.

imBunyip, 2023-03-10:
yea i  know i can get two along the chain with nai high low on each different device ,sadly im only testing with two to eventually have more along the chain.

im away from home now and cannot hook up an analyser, if i cant get it to work soon ill do some tests when home.
i uncommented  `uint32_t gconf = motors[motorCount].readRegister(TMC5160_Reg::GCONF, &readStatus);.`
because at first it was only able to find one motor,

im getting these warnings, when trying to compile the original code above with uint32_t gconf line uncommented.

```
38:12: warning: enumeration value 'INVALID_FORMAT' not handled in switch [-Wswitch]
     switch (readStatus) {
            ^
```
and 

```
/Users/me/Documents/Arduino/libraries/TMC5160_Arduino-master/src/TMC5160_UART.cpp: In member function 'uint32_t TMC5160_UART_Generic::readRegister(uint8_t, TMC5160_UART_Generic::ReadStatus*)':
/Users/me/Documents/Arduino/libraries/TMC5160_Arduino-master/src/TMC5160_UART.cpp:55:13: warning: 'readStatus' may be used uninitialized in this function [-Wmaybe-uninitialized]
  ReadStatus readStatus;
             ^
```

from what i read online even tho these are warnings if it is not initialized then it will output undefined possible random behaviour.

in the printed errors im also at times getting bad crc, which if true would mean whilst in the readstatus switch it should call 
` case TMC5160_UART::BAD_CRC:`
yet it never has.

the nai high low on the first motors is working,i have also individually tested the switches manually.

so far i have not changed anything from what you can see above, in the schematic and the code.
my hunch is that the switch is not working because of the readstatus warnings.

i understand its hard to debug something of someone else,but your already helping loads,

thanks

tommag, 2023-03-10:
Mmmh reading `gconf` (or any register for that matter) is NOT OPTIONAL if you want `readStatus` to have a meaningful value... 

Have a look again at this code : 

```
TMC5160_UART::ReadStatus readStatus;
uint32_t gconf = motors[motorCount].readRegister(TMC5160_Reg::GCONF, &readStatus);
switch (readStatus) {...
```

If you don't execute `readRegister` then readStatus will never change from its initialization value, which is SUCCESS. 

The important bit for motor detection is the result of this initial read, either it returns SUCCESS which means there is a motor replying, or any other status code which means there isn't. By commenting this line you're just skipping the test. 

Don't worry about the warnings they're not important here.

imBunyip, 2023-03-10:
ok so running the code with it like this,
```
TMC5160_UART::ReadStatus readStatus;
uint32_t gconf = motors[motorCount].readRegister(TMC5160_Reg::GCONF, &readStatus);
switch (readStatus) {...
```

it only finds one motor now, the motor does enable however ,but it is no longer able to switch nao i.e IO_INPUT_OUTPUT ,and therefore doesnt find the next motor.
the motor also no longer moves now.

it outputs this 
```
Starting to look for TMC5160 

Read 0x2: 0x0
Read 0x0: 0x9
Found TMC5130 @ address 
Writing 0x3: 0x2FD
Read 0x2: No reply (0 bytes read)
{0x1 0x0 0x0 0x0 0x90 0x11 0xFF 0x1F }
Resetting communication.
```
and it finds the new motor at address 254 etc,

tommag, 2023-03-10:
Now this is looking much better.

Check out my first comment https://github.com/tommag/TMC5160_Arduino/issues/24#issuecomment-1448898319 
I'm pretty sure you haven't changed `setSlaveAddress` with false to take into account that NAI is low and not high. Also you can't use address 254 with NAI low so you should start at 253 (check description of register 0x03 in the datasheet)

imBunyip, 2023-03-10:
ok so im using this to move motors.
```
    motors[0].setTargetPosition(dir ? 200 : 0);   // 1 full rotation = 200s/rev
    motors[1].setTargetPosition(dir ? 2000 : 0);  // 1 full rotation = 200s/rev
```
changing from 254 to 253 acts as so
`uint8_t address = 254 - motorCount; // using this both motors move as tho there motors[1] i.e back and forth between 2000 and 0`
to
`uint8_t address = 253 - motorCount; // using this it finds 2 motors and the first motor moves as expected. i.e motors[0]  the second does not move`

seems to be working better but the second motor does not move ,to (dir ? 2000 : 0) as it should first motor seems correct but im still getting allot of no reply from feedback.

yes originaly
`void setSlaveAddress(uint8_t slaveAddress, bool NAI=true); `
is defintaly set as false within TMC5160.h like so
`void setSlaveAddress(uint8_t slaveAddress, bool NAI=false);` 
and also like so 
`motors[motorCount].setSlaveAddress(address,` false);

tommag, 2023-03-10:
Please post the log of the init as you did before, otherwise I can't give you any advice.

imBunyip, 2023-03-10:
full log here,
finds both motors ,motor  0 moves back and forth 200 and 0 ,
motor 1 does not move.
tarting to look for TMC5160 

Read 0x2: 0x0
Read 0x0: 0x9
Found TMC5130 @ address 
Writing 0x3: 0x2FD
Read 0x2: No reply (0 bytes read)
{0x1 0x0 0x0 0x0 0x90 0x11 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFD 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFD 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
Writing 0x3: 0x2FD
Read 0x2: No reply (0 bytes read)
{0x1 0x0 0x0 0x0 0x90 0x11 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFD 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFD 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
Writing 0x3: 0x2FD
Read 0x2: No reply (0 bytes read)
{0x1 0x0 0x0 0x0 0x90 0x11 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFD 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFD 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
New address : 253

Writing 0x1: 0x5
Read 0x2: 0x2
Writing 0x1: 0x5
Read 0x2: 0x3
Writing 0xA: 0x80400
Read 0x2: 0x4
Writing 0xB: 0x28
Read 0x2: 0x5
Writing 0x10: 0x70E0C
Read 0x2: 0x6
Writing 0x70: 0xC409001E
Read 0x2: 0x7
Writing 0x70: 0xC40D001E
Read 0x2: 0x8
Writing 0x6C: 0x10045
Read 0x2: 0x9
Writing 0x20: 0x0
Read 0x2: 0xA
Writing 0x0: 0x4
Read 0x2: 0xB
Writing 0x23: 0x0
Read 0x2: 0xC
Writing 0x2B: 0x23
Read 0x2: 0xD
Writing 0x25: 0x0
Read 0x2: 0xE
Writing 0x20: 0x0
Read 0x2: 0xF
Writing 0x27: 0x4EA4A
Read 0x2: 0x10
Writing 0x26: 0xF45
Read 0x2: 0x11
Writing 0x28: 0xF45
Read 0x2: 0x12
Writing 0x24: 0xF45
Read 0x2: 0x13
Writing 0x2A: 0xF45
Read 0x2: 0x14
Writing 0x4: 0x0
Read 0x2: 0x15
Resetting communication.
Read 0x2: 0x0
Read 0x0: 0x9
Found TMC5130 @ address 
Writing 0x3: 0x2FC
Read 0x2: No reply (0 bytes read)
{0x1 0x0 0x0 0x0 0x90 0x11 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFC 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFC 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
Writing 0x3: 0x2FC
Read 0x2: No reply (0 bytes read)
{0x1 0x0 0x0 0x0 0x90 0x11 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFC 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFC 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
Writing 0x3: 0x2FC
Read 0x2: No reply (0 bytes read)
{0x1 0x0 0x0 0x0 0x90 0x11 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFC 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
Read 0x2: No reply (0 bytes read)
{0xFC 0x2 0x0 0x0 0x34 0x7 0xFF 0x1F }
Resetting communication.
New address : 252

Writing 0x1: 0x5
Read 0x2: 0x2
Writing 0x1: 0x5
Read 0x2: 0x3
Writing 0xA: 0x80400
Read 0x2: 0x4
Writing 0xB: 0x28
Read 0x2: 0x5
Writing 0x10: 0x70E0C
Read 0x2: 0x6
Writing 0x70: 0xC409001E
Read 0x2: 0x7
Writing 0x70: 0xC40D001E
Read 0x2: 0x8
Writing 0x6C: 0x10045
Read 0x2: 0x9
Writing 0x20: 0x0
Read 0x2: 0xA
Writing 0x0: 0x4
Read 0x2: 0xB
Writing 0x23: 0x0
Read 0x2: 0xC
Writing 0x2B: 0x23
Read 0x2: 0xD
Writing 0x25: 0x0
Read 0x2: 0xE
Writing 0x20: 0x0
Read 0x2: 0xF
Writing 0x27: 0x4EA4A
Read 0x2: 0x10
Writing 0x26: 0xF45
Read 0x2: 0x11
Writing 0x28: 0xF45
Read 0x2: 0x12
Writing 0x24: 0xF45
Read 0x2: 0x13
Writing 0x2A: 0xF45
Read 0x2: 0x14
Writing 0x4: 0x0
Read 0x2: No reply (0 bytes read)
{0x1 0x0 0x0 0x0 0x90 0x11 0xFF 0x1F }
Resetting communication.
Read 0x2: 0x15
Resetting communication.
Starting up
Writing 0x2D: 0xC800
Read 0x2: No reply (0 bytes read)
{0x1 0x0 0x0 0x0 0x90 0x11 0xFF 0x1F }
Resetting communication.
Read 0x2: 0x16
Writing 0x2D: 0x7D000
Read 0x2: No reply (0 bytes read)
{0x1 0x0 0x0 0x0 0x90 0x11 0xFF 0x1F }
Resetting communication.
Read 0x2: 0x15
Writing 0x2D: 0x7D000
Read 0x2: No reply (0 bytes read)
{0x1 0x0 0x0 0x0 0x90 0x11 0xFF 0x1F }
Resetting communication.

tommag, 2023-03-10:
It looks like communication to motor 1 is working well for init (you can see `Read 0x2` incrementing, this is the write counter indicating that registers are successfully written. Then after motor 0 has started, it doesn't work so well (Read 0x2 returns 0x15 instead of 0x16 which means that the write to 0x2D = XTARGET has failed). 

Maybe an interference issue on the bus when the first motor starts to run ? 
Try to set motor 1 position before motor 0 to see if it changes anything ?

I won't be able to help you any further, I think you'll need a logic analyzer / oscilloscope on the bus to go further.

imBunyip, 2023-03-10:
yep, setting motor 1 position before the 0 makes motor 1 work going between 0 and 2000.
again the other motor 0 does not.

could be interference.the pcb is very compact, im currently chaining them with an aux to aux cable.

thanks for your time , ill get out an analyzer when home and possible be back here.

tommag, 2023-03-20:
Great. You can try to lower further the baudrate for a more robust communication, or just get away with it.

Also if you want to change parameters regularly (like updating target positions frequently) you may want to switch to STREAMING_MODE instead of RELIABLE_MODE (does not check for 0x2 = the transmission counter after each write - suitable for cases where you update regularly every register).

closed 2023-03-20 by tommag