tommag/TMC5160_Arduino #31: Set target function

simonedallai, 2024-03-18:
Hi, I would like to use the set target function in reference not to the position of the steps but in relation to the encoder, how can I do it?

tommag, 2024-03-19:
I'm not sure I understand what you want to do, but you should be able to achieve what you want using `getEncoderPosition()` and `getCurrentPosition()`? 

There is no builtin closed loop management inside the TMC5160. You have to handle it yourself.

closed 2024-03-19 by tommag

simonedallai, 2024-03-19:
ok if I haven't misunderstood the set target function is only in relation
to the microsteps, if I want a closed loop I have to check the position of
the encoder at all times......

tommag, 2024-03-20:
Exactly. TMC4361 has built-in closed loop support but not TMC5160.

simonedallai, 2024-03-20:
but in the 4361 library you made, a 5160 could fit, can the two go together?