tommag/TMC5160_Arduino #35: TMC5160T PRO with AdafruitFeather M0 -> Stepper is not moving

StefanMeGit, 2024-08-22:
Hi, i try to play around with a NEMA34 and the BIGTREETECH TMC5160T Pro board. It looks like i hooked up evrything correctly but i still cant get the motor spinning. I cant turn the shaft of the motor, but it will never move from program.

BOARD: [BIGTREETECH TMC5160T Pro](https://github.com/bigtreetech/BIGTREETECH-Stepper-Motor-Driver/blob/master/TMC5160T%20Plus/Hardware/TMC5160T%20Plus-Pin.png)

Is there any hint u can give me? 

It looks like the SPI connection is just fine. The config works exexpt the motor isnt spinning...
```
Found a TMC device.
IC version: 0x30 (TMC5160).

Global status: 0x5
Driver status: 0x80080000

Start power stage tuning ? 
y/n (Default: n):
Starting operation with lowest current setting possible.

You should monitor one of the bridge outputs (motor coil connections) using a scope.
Let's choose the MOSFET gate driver current. This should be the lowest setting giving slopes < 100ns
Current DRVSTRENGTH value: 2. Press +/- to increase / decrease or d when done.+/-/d):
```

Output on the test program, stepper is not moving:
```
current position : 200.00	current speed : 0.00
current position : 200.00	current speed : 0.00
current position : 200.00	current speed : 0.00
current position : 199.36	current speed : -25.23
current position : 194.21	current speed : -76.13
current position : 183.86	current speed : -127.04
current position : 168.34	current speed : -177.94
current position : 147.63	current speed : -228.85
current position : 121.73	current speed : -279.75
current position : 91.07	current speed : -301.61
current position : 62.94	current speed : -250.73
current position : 39.99	current speed : -199.84
current position : 22.23	current speed : -148.96
current position : 9.64	current speed : -98.08
current position : 2.24	current speed : -47.22
current position : 0.00	current speed : 0.00
current position : 0.00	current speed : 0.00
current position : 0.00	current speed : 0.00
current position : 0.00	current speed : 0.00
current position : 0.00	current speed : 0.00
current position : 0.00	current speed : 0.00
current position : 0.00	current speed : 0.00
current position : 0.00	current speed : 0.00
current position : 0.00	current speed : 0.00
current position : 0.00	current speed : 0.00
current position : 0.00	current speed : 0.00
current position : 0.00	current speed : 0.00
```

tommag, 2024-08-23:
Hello, 
You're talking about TMC5160T PRO but sent a link for TMC5160T PLUS, which one is it ? 

Anyway, these lines should give some good hints : 
> Global status: 0x5
> Driver status: 0x80080000

According to the TMC5160 datasheet (that you definitely need to know a little bit to use this board) : 
GSTAT = 0x5 => charge pump undervoltage

So it looks like the charge pump undervoltage should be investigated : how is the board powered, is the voltage high enough, the power supply strong enough ?

I strongly suggest that you run the "Config wizard" example and follow the instructions in order to define the correct parameters for your driver board and motor, there are a lot of parameters and I tried to make it a bit more user friendly with this code.

tommag, 2024-08-23:
Looking at the TMC5160T PLUS schematic it seems that you need to provide an external 24V supply as well as the 8-60V motor supply and the VIO voltage. Is it the case ?

StefanMeGit, 2024-08-23:
Yes its the case. I provided VIO, 24 and 24V for the motor.

StefanMeGit, 2024-08-27:
The Problem was the SPI/STANDALONE/UART mode jumper. The shematics are simply wrong.

For SPI u need to have to bridge the bottom two pins. The picture shows it wrong...

[image]

closed 2024-09-02 by tommag