tommag/TMC5160_Arduino #43: Tmc

budabalint, 2025-08-08:
I have a TMC5160 stepper driver that I’m trying to get working.
The goal is to control it only via SPI, but I can’t get the driver to start up at all.
I’ve tried both SPI control and Step/Dir signals, as well as several C++ libraries, but nothing seems to work.

This is the official github repository for the panel:
https://github.com/FYSETC/FYSETC-QHV5160

Most recently, I tested with this library but the driver still didn’t respond.
I also have a TMC2209 driver, and with that one my motor runs perfectly over UART, so I know the motor itself is fine.

With the TMC5160, however, the motor doesn’t move at all — it doesn’t even hold torque.
Using a multimeter, I confirmed that it’s receiving the correct voltages (3.3 V and 16 V), GND is common everywhere, and as far as I can tell the pins are wired correctly.
It just refuses to start in any way.

I’m starting to suspect a hardware fault, but I’m not 100% sure.
Does anyone have any tips or things I should check?

Here is some images of the current wiring.

[image]
[image]
[image]
[image]
[image]
[image]

Thanks in advance to anyone who can help.

budabalint, 2025-08-08:
Aham értem ez elég tág megfogalmazás hogy a chip alatt le van kötve földre a lába. Pontosan melyik lábra gondolsz és honnan tudod már ha szabad érdeklődnöm? Illetve miért árúlná.k ha gyárilag le lene kötve a lába?
Köszi a választ!

budabalint, 2025-08-08:
> A chip alatt levan kotve foldre a laba, eselytelen, vegyel bob verziotAz iPhone-omról küldve2025. aug. 8. dátummal, 17:01 időpontban Buda Bálint ***@***.***> írta:budabalint created an issue ([tommag/TMC5160_Arduino#43](https://github.com/tommag/TMC5160_Arduino/issues/43))
> I have a TMC5160 stepper driver that I’m trying to get working.
> The goal is to control it only via SPI, but I can’t get the driver to start up at all.
> I’ve tried both SPI control and Step/Dir signals, as well as several C++ libraries, but nothing seems to work.
> This is the official github repository for the panel:
> https://github.com/FYSETC/FYSETC-QHV5160
> Most recently, I tested with this library but the driver still didn’t respond.
> I also have a TMC2209 driver, and with that one my motor runs perfectly over UART, so I know the motor itself is fine.
> With the TMC5160, however, the motor doesn’t move at all — it doesn’t even hold torque.
> Using a multimeter, I confirmed that it’s receiving the correct voltages (3.3 V and 16 V), GND is common everywhere, and as far as I can tell the pins are wired correctly.
> It just refuses to start in any way.
> I’m starting to suspect a hardware fault, but I’m not 100% sure.
> Does anyone have any tips or things I should check?
> Here is some images of the current wiring.
> 20250808_151653.jpg (view on web)
> 20250808_151649.jpg (view on web)
> 20250808_151640.jpg (view on web)
> 20250808_151634.jpg (view on web)
> 20250808_151644.jpg (view on web)
> 20250808_151630.jpg (view on web)
> Thanks in advance to anyone who can help.

Aham értem ez elég tág megfogalmazás hogy a chip alatt le van kötve földre a lába. Pontosan melyik lábra gondolsz és honnan tudod már ha szabad érdeklődnöm? Mert a kapcsolás alapján SPI módban van. Vagy van ilyen paneled? Illetve miért árúlnák ha gyárilag le lene kötve a lába?
Köszi a választ!

puredirect, 2025-08-08:
3 ève nekem ez volt a problèmam egy tmc chippel, 3d nyomtatohoz gyartottak es talan az spi enable vagy hasonlo cèlú làb le volt húzva földre. Nem kezdem el nèzni a te chiped, csak ötletkènt irtam hogy nezz utana, nekem ez volt a bajomAz iPhone-omról küldve

https://github.com/FYSETC/FYSETC-QHV5160